LAStools/LAStools #230: LAStools master can't be built as DLL (BUILD_SHARED_LIBS) for Windows anymore

shintadono, 2025-01-15:
While trying to build LAStools current master as BUILD_SHARED_LIBS, I noticed quite a few link issues.
```
error LNK2019: unresolved external symbol "void __cdecl byebye(void)" (?byebye@@YAXXZ)
error LNK2019: unresolved external symbol "void * __cdecl realloc_las(void *,unsigned __int64)" (?realloc_las@@YAPEAXPEAX_K@Z)
error LNK2001: unresolved external symbol "long * lasmessage_cnt" (?lasmessage_cnt@@3PAJA)
error LNK2001: unresolved external symbol "bool wait_on_exit" (?wait_on_exit@@3_NA)
error LNK2001: unresolved external symbol "bool halt_on_error" (?halt_on_error@@3_NA)
...
```
Most of them seem to originate in the mydefs.hpp, but I haven't checked everything.

For me it looks like there is some DLL export declarations missing.

There are also 3 boolean variables and a long array defined as extern - this might not even work with DLLs. IIRC DLL can only export functions/methods, but not variables. To make that work, would need to have getters and setters for those bools.

We would like to use LASlib as DLL (and as SO on Linux - haven't tested that yet), and building the LASlib part of the solution actually creates a DLL file (though the install build target fails, because the exe files are not there). But when trying to include the headers in our code, all hell break loose.

rapidlasso, 2025-01-22:
We did not maintain the BUILD_SHARED_LIBS build option recently. We will fix this some day, but it has not much priority for us. If you want to do a favor to the community just fix this BUILD_SHARED_LIBS build and post a pull request and we will merge immediately.Thanks!

shintadono, 2025-03-31:
Thanks you so much! <3