LAStools/LAStools #232: Problem with LASMerge

rakyuv, 2025-02-04:
Hi, 

I am trying to merge two adjacent LAZ files (two tiles containing terrain information which are next to each other geographically). I am using it in MacOS. So I followed https://rapidlasso.de/using-lastools-on-mac-os-x-with-wine/, to install and use it. I use LAStools in QGIS. 

When I merged the files with Lasmerge, I get the following log:

Version de QGIS : 3.28.1-Firenze
Révision du code : fde3b8fbb8
Version de Qt : 5.15.2
Version de Python : 3.9.5
Version de GDAL : 3.3.2
Version de GEOS : 3.9.1-CAPI-1.14.2
Version de Proj : Rel. 8.1.1, September 1st, 2021
Version de PDAL : 2.3.0 (git-version: Release)
Algorithme commencé à: 2025-02-04T11:56:04
Démarrage de l'algorithme 'lasmerge (file)'…
Paramètres en entrée:
{ 'ADDITIONAL_OPTIONS' : '', 'APPLY_FILE_SOURCE_ID' : False, 'CPU64' : True, 'FILE2' : '/Users/user/Downloads/LHD_FXX_0849_6515_PTS_O_LAMB93_IGN69.copc.laz', 'FILE3' : '', 'FILE4' : '', 'FILE5' : '', 'FILE6' : '', 'FILE7' : '', 'FILES_ARE_FLIGHTLINES' : False, 'GUI' : False, 'INPUT_LASLAZ' : '/Users/user/Downloads/LHD_FXX_0848_6515_PTS_O_LAMB93_IGN69.copc.laz', 'VERBOSE' : False }

LAStools command line
/opt/homebrew/bin/wine /Users/user/Downloads/LAStools/bin/lasmerge64.exe -i "/Users/user/Downloads/LHD_FXX_0848_6515_PTS_O_LAMB93_IGN69.copc.laz" -i /Users/user/Downloads/LHD_FXX_0849_6515_PTS_O_LAMB93_IGN69.copc.laz -cpu64
LAStools console output
[mvk-info] MoltenVK version 1.2.10, supporting Vulkan version 1.2.290.
The following 110 Vulkan extensions are supported:
VK_KHR_16bit_storage v1
VK_KHR_8bit_storage v1
VK_KHR_bind_memory2 v1
VK_KHR_buffer_device_address v1
VK_KHR_calibrated_timestamps v1
VK_KHR_copy_commands2 v1
VK_KHR_create_renderpass2 v1
VK_KHR_dedicated_allocation v3
VK_KHR_deferred_host_operations v4
VK_KHR_depth_stencil_resolve v1
VK_KHR_descriptor_update_template v1
VK_KHR_device_group v4
VK_KHR_device_group_creation v1
VK_KHR_driver_properties v1
VK_KHR_dynamic_rendering v1
VK_KHR_external_fence v1
VK_KHR_external_fence_capabilities v1
VK_KHR_external_memory v1
VK_KHR_external_memory_capabilities v1
VK_KHR_external_semaphore v1
VK_KHR_external_semaphore_capabilities v1
VK_KHR_fragment_shader_barycentric v1
VK_KHR_format_feature_flags2 v2
VK_KHR_get_memory_requirements2 v1
VK_KHR_get_physical_device_properties2 v2
VK_KHR_get_surface_capabilities2 v1
VK_KHR_imageless_framebuffer v1
VK_KHR_image_format_list v1
VK_KHR_incremental_present v2
VK_KHR_maintenance1 v2
VK_KHR_maintenance2 v1
VK_KHR_maintenance3 v1
VK_KHR_map_memory2 v1
VK_KHR_multiview v1
VK_KHR_portability_subset v1
VK_KHR_push_descriptor v2
VK_KHR_relaxed_block_layout v1
VK_KHR_sampler_mirror_clamp_to_edge v3
VK_KHR_sampler_ycbcr_conversion v14
VK_KHR_separate_depth_stencil_layouts v1
VK_KHR_shader_draw_parameters v1
VK_KHR_shader_float_controls v4
VK_KHR_shader_float16_int8 v1
VK_KHR_shader_integer_dot_product v1
VK_KHR_shader_non_semantic_info v1
VK_KHR_shader_subgroup_extended_types v1
[... 117 lines cut ...]
00d4:err:hid:handle_DeviceMatchingCallback Ignoring HID device 0x7fd9cc704fa0 (vid 05ac, pid 0343): not a joystick or gamepad
00d4:err:hid:handle_DeviceMatchingCallback Ignoring HID device 0x7fd9cc7062b0 (vid 05ac, pid 8104): not a joystick or gamepad
00d4:err:hid:handle_DeviceMatchingCallback Ignoring HID device 0x7fd9cc7048c0 (vid 004c, pid 0269): not a joystick or gamepad
00d4:err:hid:handle_DeviceMatchingCallback Ignoring HID device 0x7fd9cc705940 (vid 0000, pid 0000): not a joystick or gamepad
00b0:err:ntoskrnl:ZwLoadDriver failed to create driver L"\\Registry\\Machine\\System\\CurrentControlSet\\Services\\winebth": c00000e5
003c:fixme:service:scmdatabase_autostart_services Auto-start service L"winebth" failed to start: 1359
ERROR: cannot open 'file_list_buffered.14836340772344663953.txt' 

Execution completed in 5.93 secondes
Résultats :
{'commands': ['/opt/homebrew/bin/wine '
'/Users/user/Downloads/LAStools/bin/lasmerge64.exe',
'-i',
'"/Users/user/Downloads/LHD_FXX_0848_6515_PTS_O_LAMB93_IGN69.copc.laz"',
'-i',
'/Users/user/Downloads/LHD_FXX_0849_6515_PTS_O_LAMB93_IGN69.copc.laz',
'-cpu64']}

Chargement des couches de résultat
Algorithme 'lasmerge (file)' terminé

Could you please tell me how I can correct this error, while merging the files ?

Thanks in advance,
Rakesh

kdjacob2000, 2025-03-14:
I have the same problem. Same installation as above.

PLEASE HELP!

kdjacob2000, 2025-03-14:
I will try this but before I do to be sure I understand:

Can I assume you are using Terminal for this command?

Also, do you have to change the directory (cd in Terminal) to where the .laz files are to run this?

Thanks for your help!

Ken

kdjacob2000, 2025-03-15:
Hi rakyuv,

I was able to get this to work from Terminal.

Thank you for this valuable workaround.

I hope someone picks up this bug in using lasmerge from QGIS and fixes it.

K

rakyuv, 2025-03-17:
Hi @kdjacob2000,

Sorry for my delayed answer. I'm glad you were able to make it work with command line. 

Indeed, I'll be happy if the the bug gets fixed in lasmerge in QGIS. 

Best regards,
R

BB-Heidelberg, 2025-03-17:
Hi R, there is no "lasmerge in QGIS". There is a lasmerge - and this runs in the terminal quite well. There is a LAStools plugin for QGIS, wrapping lasmerge to be able to be used by QGIS. And there is wine, to enable to run windows programs on mac.
lasmerge is running, the LAStools plugin is running on Windows and Linux systems - so the problem above seems to be about wine.
Where is the bug to get fixed on lasmerge?

rakyuv, 2025-03-17:
Hi @BB-Heidelberg . My apologies. As you mentioned, lasmerge works perfectly in terminal. 

Like me and K, some people tried to use LAStools with QGIS first, and ran into some problems like:
```
003c:fixme:service:scmdatabase_autostart_services Auto-start service L"winebth" failed to start: 1359
ERROR: cannot open 'file_list_buffered.14836340772344663953.txt'
```

I'll check with QGIS on how to make it work. I'll close this issue for the moment. 

And even if someone runs into similar problems with QGIS in the future, I'm sure this post could serve them to point out that it is much easier to use LAStools with terminal, than with QGIS. 

Best regards,
Rakesh

closed 2025-03-17 by rakyuv

BB-Heidelberg, 2025-03-17:
perfect, thanks Rakesh